vitiral/gpio #6: set() and output() always return 1 when using gpio.HIGH and gpio.LOW

ghuband, 2017-10-03:
# python
Python 2.7.11 (default, Feb 15 2017, 21:16:35) 
[GCC 5.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import gpio
>>> ch="43"
>>> gpio.setup(ch,gpio.OUT)
DEBUG:gpio:Setup 43: out
DEBUG:gpio:writing: <open file '/sys/class/gpio/gpio43/direction', mode 'w+' at 0x769da180>: out
DEBUG:gpio:Write 43: 0
DEBUG:gpio:writing: <open file '/sys/class/gpio/gpio43/value', mode 'w+' at 0x769da128>: 0
>>> gpio.set(ch,gpio.HIGH)
DEBUG:gpio:Write 43: 1
DEBUG:gpio:writing: <open file '/sys/class/gpio/gpio43/value', mode 'w+' at 0x769da128>: 1
>>> gpio.set(ch,gpio.LOW)
DEBUG:gpio:Write 43: 1
DEBUG:gpio:writing: <open file '/sys/class/gpio/gpio43/value', mode 'w+' at 0x769da128>: 1

Using True and False works correctly.

vitiral, 2017-10-03:
what do you mean "return 1" -- do you mean the pin value is 1? `set()` doesn't return anything...

Set is defined as:
```
def set(pin, value):
    '''set the pin value to 0 or 1'''
    value = int(bool(value))
```

Both `gpio.IN` and `gpio.OUT` are "truthy" in python, which is why it is always one.

There could be an assertion that makes sure the value is 0 or 1 instead of the `int(bool(value))` that is going on.

ghuband, 2017-10-03:
From the python raspberry pi gpio documentation:
https://sourceforge.net/p/raspberry-gpio-python/wiki/BasicUsage/

Output
To set the output state of a GPIO pin:
GPIO.output(channel, state)
(where channel is the channel number based on the numbering system you have specified (BOARD or BCM)).
State can be 0 / GPIO.LOW / False or 1 / GPIO.HIGH / True.

Your set() and output() functions do not work correctly when using gpio.HIGH and gpio.LOW.  At least document that these functions only work when using True or False.

vitiral, 2017-10-03:
oh, I misread. Thanks for the bug and sorry about the confusion.

commit 9f620079 (2017-10-03): fix #6: set(pin, gpio.LOW) == set(pin, 0)
closed 2017-10-03 by vitiral