vitiral/gpio #7: cleanup(pin) does not remove pin file from /sys/class/gpio

ghuband, 2017-10-03:
```
# python
Python 2.7.11 (default, Feb 15 2017, 21:16:35) 
[GCC 5.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import gpio
>>> ch="43"    
>>> gpio.setup(ch,gpio.OUT)
DEBUG:gpio:Creating Pin 43
DEBUG:gpio:writing: <open file '/sys/class/gpio/export', mode 'w' at 0x769b8128>: 43
DEBUG:gpio:Setup 43: out
DEBUG:gpio:writing: <open file '/sys/class/gpio/gpio43/direction', mode 'w+' at 0x769b81d8>: out
DEBUG:gpio:Write 43: 0
DEBUG:gpio:writing: <open file '/sys/class/gpio/gpio43/value', mode 'w+' at 0x769b8180>: 0
>>> gpio.cleanup(ch)
>>> quit()
# ls /sys/class/gpio
export  gpio43  gpiochip0  gpiochip128  gpiochip160  gpiochip192  gpiochip32  gpiochip64  gpiochip96  unexport
```

vitiral, 2017-10-03:
Thanks for the issue. Can you link to the reference which says that the pin file *should* be removed? It's been several years since I've used this library.

sheffieldnick, 2017-10-15:
The pin file (actually a directory) is removed by the /sys/class/gpio/unexport operation. At least it is for me. If unexport was being called for @ghuband then they should see a DEBUG:gpio:Unexporting pin {43}, which they aren't?

I'm not a python programmer, but I wondered if there was an issue in cleanup() because ch has been defined as a string "43" instead of ch=43 ?

vitiral, 2017-10-15:
yes, pin being defined as a string is very likely the problem.

We should probably *at least* assert that the type of pin is an int. We might even want to raise a KeyError if the pin isn't found, although that would break the current API.

vitiral, 2017-10-15:
@ghuband please confirm whether changing pin to an `int` fixes your issue. If it does, I will add that assertion and close this

vitiral, 2017-10-15:
labeling this as an enhancement (to add the `TypeError`) since it looks like that is the underlying issue.

sheffieldnick, 2017-10-16:
Every other function in the library uses _verify() to silently cast the pin argument to int, without raising an error, so for consistency would it make sense for cleanup() to also cast to an int, rather than giving a type error if the argument isn't an int?

vitiral, 2017-10-21:
agh... I think we should remove that functionality from `_verify()` instead... (among other things, that is a tiny performance hit -- it is also just annoying)

vitiral, 2017-10-21:
I was so bad at programming when I first wrote this library, I feel really bad for anyone who depended on that being a string ☹️

Gadgetoid, 2021-11-03:
Closing this, since it's more than likely to have been fixed in my suite of changes. Please open a new issue for future discussion.

closed 2021-11-03 by Gadgetoid