vitiral/gpio #8: cleanup() function that cleans up all open pins for RPi.GPIO compatibility

sheffieldnick, 2017-10-15:
I noticed that the cleanup(pin) only cleans that single pin, which isn't compatible with the RPi.GPIO library that has a zero-argument cleanup(). This might be a suitable addition if you need such a function?

`# RPi.GPIO compatible cleanup()
def cleanup():

    for pin in _open:
        files = _open[pin]
        files['value'].close()
        files['direction'].close()
        if os.path.exists(gpiopath(pin)):
            log.debug("Unexporting pin {0}".format(pin))
            with _export_lock:
                with open(pjoin(gpio_root, 'unexport'), 'w') as f:
                    _write(f, pin)
    _open.clear()`

vitiral, 2017-10-15:
I vaguly remember purposely not supporting that functionality since it really breaks down the "compartmentalization" of different parts of your program. I generally dislike things that affect global state.

 I could make pin a default of None, in which case it cleans up everything. The question is should we do a `ls` to find *all pins* that exist and clean up all of them, or only the ones in `_open`?

sheffieldnick, 2017-10-15:
I think the RPi.GPIO library **only** cleans pins that have been opened by itself, so as not to interfere with other processes, etc?
[http://raspi.tv/2013/rpi-gpio-basics-3-how-to-exit-gpio-programs-cleanly-avoid-warnings-and-protect-your-pi](url)

vitiral, 2017-10-15:
sounds good

vitiral, 2017-10-15:
I feel like there was an oversight here: `cleanup` should also remove the pin from `_open` correct?

sheffieldnick, 2017-10-15:
Yes, it should! Ooops

commit d92bcb52 (2017-10-15): fix #8: RPi.GPIO compatibility for cleanup() and use state object.
commit 3bca3c29 (2017-10-21): Merge pull request #9 from vitiral/issue_8

fix #8: RPi.GPIO compatibility for cleanup()
closed 2017-10-21 by vitiral