vitiral/gpio #10: setup() function that handles a list of pins for RPi.GPIO compatibility

sheffieldnick, 2017-10-16:
RPi.GPIO.setup() appears to take either a single pin (int) as its first argument, or a list of integer pins. e.g.,

`GPIO.setup([DAT,CLK],GPIO.OUT)`

the current gpio.setup() only copes with a single integer pin. Something like this code should fix the incompatibility?

```# RPi.GPIO compatible setup()
# pin argument can be int or list of int
def setup(pin, mode, pullup=None, initial=False):
    if isinstance(pin, list):
        for apin in pin:
            setup(apin, mode, pullup, initial)
        return
# ... existing single pin code
```

sheffieldnick, 2017-10-16:
The @_verify decorator stops this change from working, as it doesn't accept 'pin' as a list. Could _verify be called after checking whether pin is a list or not?

sheffieldnick, 2017-10-16:
RPi.GPIO also lets cleanup() and output() take an optional list of integer pin numbers.

vitiral, 2017-10-16:
I am really hesitant to support this functionality. Nothing stops the user from doing the same loop or creating a function to do so.

Also, `isinstance(pin, list)` is not sufficient. You should also allow tuples, so `isinstance(pin, (list, tuple))` you might think... well, we probably want to support iterators in general, so now it becomes 

```
if hasattr(pin, '__iter__'):
    for p in pin:
        setup(p, mode, pullup, initial)
```

And I *really* don't like this

sheffieldnick, 2017-10-16:
Fair enough. If compatibility isn't the main goal, then then at least it would be good to have some documentation showing where the gpio functions differ from the RPi.GPIO library? Just to give a heads-up to anyone trying to port some code that was written for RPi.GPIO?

vitiral, 2017-10-21:
I would put the docs in the README. Again, PR's welcome.

rahulraghu94, 2017-11-02:
Thinking out loud:

Can we support ONLY lists or tuples?
We can include this in the docs, saying that only lists or tuples are supported.

```
@_verify
def _setup(pin, mode=None, pullup=None, initial=False):
    '''Setup pin with mode IN or OUT.

    Args:
        pin (int):
        mode (str): use either gpio.OUT or gpio.IN
        pullup (None): rpio compatibility. If anything but None, raises
            value Error
        pullup (bool, optional): Initial pin value. Default is False
    '''
    if pullup is not None:
        raise ValueError("sysfs does not support pullups")

    if mode not in (IN, OUT, LOW, HIGH):
        raise ValueError(mode)

    log.debug("Setup {0}: {1}".format(pin, mode))
    f = _open[pin].direction
    _write(f, mode)
    if mode == OUT:
        if initial:
            set(pin, 1)
        else:
            set(pin, 0)

def setup(pins, mode=None):
    if isinstance(pins, (list, tuple)):
        for pin in pins:
            _setup(pin, mode)
    else:
        _setup(pins, mode)
```

I have a test branch with suggested implementation here: https://github.com/rahulraghu94/gpio/blob/test-list-setup/gpio.py#L144

Gadgetoid, 2021-08-12:
Used `type(pin) in (list, tuple)` but I've added support for lists in `cleanup` and `setup` in #22.

I know I'm a day late or a dollar short, but if you somehow still care enough to test these changes it would be appreciated.

vitiral, 2021-08-12:
you could use `isinstance(value, collections.Iterable)` and support pretty much anything.

Gadgetoid, 2021-11-03:
A pre-release test version of this these improvements can now be found on testpypi here: https://test.pypi.org/project/gpio-test/1.0.0/

Closing this issue to clean house. Please open a new one if this functionality isn't up to scratch!

closed 2021-11-03 by Gadgetoid