vitiral/gpio #11: Speed optimisations for input()/output()

sheffieldnick, 2017-10-16:
Obviously the sysfs isn't ideal if you want to waggle GPIO pins at high speed, but the library can be made a bit faster:

1. input()/output() are aliases for read()/set(), but the way they are defined _verify() is called twice for each call, once for output() and then again for set(). So change to:

```
def input(pin):
    '''read the pin. Same as read'''
    return read(pin)

def output(pin, value):
    '''set the pin. Same as set'''
    return set(pin, value)
```

or just:

```
input = read
output = set
```

2. Wrapping read()/set() with _verify() adds around 20% overhead to those calls. Is it really needed? If it is needed, doing the same "if pin in _open" test **inside** the set() function has far less overhead

3. Is there any benefit from defining LOW, HIGH = 'low', 'high' instead of as int? Removing the:

```    
    if value is LOW:
        value = 0
    value = int(bool(value))
```

from inside read()/set() gives another 5-10% improvement. The RPi.GPIO library uses ints rather than strings for pin values.

4. The logging call adds around 30% overhead to read()/set() even with no debug output (!) i.e., without "logging.basicConfig(level=logging.DEBUG)" set at the top of the library. Even this would help a lot:

```    
    if (log.level==logging.DEBUG):
        log.debug("Write {0}: {1}".format(pin, value))
```

All told, those 4 changes doubles the maximum speed of the library, on a modest dual core ARM board.

vitiral, 2017-10-21:
sounds good, I will accept a PR that does one or all of the above.

rahulraghu94, 2017-10-31:
Hello,
I'll came across this library for some personal project and I thought I could take over this issue (and the other two open issues). Would it be okay if I included all fixes to all the 3 issues in one PR or would you prefer three separate PR's?
Also, thanks for writing this library, it's proved very useful :D

vitiral, 2017-10-31:
@rahulraghu94 I welcome any PRs!

Note that this is really the only "meaty" issue.

- #10 is pretty much just docs saying we don't support that
- #7 is raise a type error if the input is a `str`, instead of silently casting to int

rahulraghu94, 2017-11-02:
Hello,

I just oticed that issue #7 (is raise a type error if the input is a str, instead of silently casting to int) is already being implemented at: https://github.com/vitiral/gpio/blob/master/gpio.py#L98

i.e., incoming pin is checked against integer. If not int, TypeError is thrown

sheffieldnick, 2019-02-28:
Shame none of these optimisations have been adopted in 16 months... :/
Here is my optimised gpio.py: [gpio.zip](https://github.com/vitiral/gpio/files/2916565/gpio.zip)

There are numerous changes, but the headlines are **3x faster GPIO writing**, and **2x faster GPIO reading**, benchmarked with python v3.5.2 on a NanoPi Fire3. Bigger improvements on a Raspberry Pi Zero (3.4x faster writing, 2.2x faster reading) compared to v0.2.0.

**Changes since gpio.py 0.2.0:**
1. replaced duplicate output()/input() functions with pointers to set()/read()
2. removed _verify wrapper on all functions except setup()
3. folded _verify inside setup(), and now raises error if pin argument isn't an int
4. changed LOW, HIGH to 0, 1 
5. folded _write() inside set(), and _read() inside read() to avoid extra function calls
6. rewrote log.debug() strings without .format() (38% gain)
7. default log.debug() is a no-op unless user wants debug logging (further 64% gain)
8. removed all type conversions in set()/_write()
9. removed type conversion and strip() in read()/_read(): _TEST strip() ISN'T NEEDED?_
10. swapped set<->output() and read<->input() since output()/input() are the RPi.GPIO compatible function names
11. bumped version to 0.2.1

vitiral, 2019-02-28:
Hey @sheffieldnick I haven't used python for embedded for years. If you would like to be a maintainer of this project, please let me know.

sheffieldnick, 2019-02-28:
Thanks for the offer, but I'd rather just contribute than maintain a project.

Gadgetoid, 2021-08-12:
@sheffieldnick are you still interested in contributing/raising a PR?

sheffieldnick, 2021-08-12:
> @sheffieldnick are you still interested in contributing/raising a PR?

No. I contributed all my code over 2 years ago, and it would be trivial for a maintainer to add my changes if they'd wanted to.

Gadgetoid, 2021-08-12:
Noted. I have raised your code in PR #21 for posterity and am working on a rewrite with due credit.

closed 2021-08-12 by Gadgetoid