vitiral/gpio #14: PermissionError

mthd98, 2018-08-27:
hi vitiyal 
i have problem can you help me to fix it 
iam using  nvidia jetson tx2 and i'm new user for Ubuntu  

PermissionError(388, 'Permission denied')
> /home/nvidia/.local/lib/python3.5/site-packages/gpio.py(75)wrapped()
-> with open(pjoin(gpio_root, 'export'), 'w') as f:

how i can fix it

vitiral, 2018-08-27:
You may need to configure your privileges. Try running your script with
"sudo", i.e. "sudo python myscript.py" and see if that works.

>

jreindel, 2019-04-17:
Running python in sudo should fix this. 
Also make sure you aren't using using any commands, like gpio.output(), before doing gpio.setup. This will also return a permission error.
I recently wrote a little tutorial for this library running on Jetson TX2 over on the nvidia devtalk site: https://devtalk.nvidia.com/default/topic/1030443/using-gpio-on-nvidia-jetson-tx2/?offset=4#5330619

rhjdvsgsgks, 2021-12-01:
hi everyone

https://github.com/vitiral/gpio/blob/fd8e954321abf2746057f7a4085ba47a5a552e60/gpio/__init__.py#L19

i found that when `fmode` is `w+` will throw a error even run with sudo

```
PermissionError: [Errno 13] Permission denied: '/sys/class/gpio/export'
```
but change it to `w` can resolve this problem

Gadgetoid, 2021-12-01:
Curious. I wonder if this is true in all cases, or if we should add a way to customise `FMODE`.

VyCOSystems, 2023-02-15:
> hi everyone
>
> https://github.com/vitiral/gpio/blob/fd8e954321abf2746057f7a4085ba47a5a552e60/gpio/__init__.py#L19
>
> i found that when `fmode` is `w+` will throw a error even run with sudo
>
> ``` [...]

same for me, running version 1.0.0.
Maybe add a way for controlling F mode per pin?

some of them only need to be output only.

thanks

vitiral, 2023-02-15:
sounds like a reasonable feature/setting to add when creating pins