vitiral/gpio #16: Use

VincentDuf, 2018-09-06:
Hi vitiral,
I'm new on TX2 and python and I'm struggling using your gpio.py: how should I call the pins in my python program ?

Thanks in advance

vitiral, 2018-09-06:
What have you tried? What error messages are you seeing? What is TX2?

You should be able to setup the pin with `setup(pin, mode)` and then set the value.

VincentDuf, 2018-09-06:
Sorry, I was talking about the Nvidia Jetson TX2,
Here is a classical error message:
```
Traceback (most recent call last):
  File "./MainAphid2.py", line 52, in <module>
    setup(Pin1,'in')
  File "/home/nvidia/gpio/gpio.py", line 76, in wrapped
    _write(f, pin)
  File "/home/nvidia/gpio/gpio.py", line 56, in _write
    f.flush()
IOError: [Errno 22] Invalid argument
```
Maybe it's coming from the pin number, should I use number between 1 and 40 or the number provided by the description document (using sysfs) ?

Also to "read" the value, is it the function read or output ?
Thanks

vitiral, 2018-09-06:
use `read` to read a value and `set` to set the value. `input/output` are the exact same but I find the names confusing (they only exist for backcompat).

You should only use `gpio.IN`, not `"in"` directly, although what you did should _technically_ work.

Sorry, I don't know what the problem is for you particular use case. Make sure your user has the appropriate permissions (you can try `sudo python <your script>` to ensure that).

VincentDuf, 2018-09-07:
Thank you for your answer, but can you tell me if I should use the pin number (1 to 40) or the LinuxGPIO number (random number like 395 or 410 etc...) ?

vitiral, 2018-09-07:
I don't know. That is very device dependent and I don't know the details of
your device, sorry.

ikoc, 2019-04-02:
with nvidia tx2 devkit this code works.
pin header : https://www.jetsonhacks.com/nvidia-jetson-tx2-j21-header-pinout/

`import gpio
import time

trig = 394
echo = 393

print "HC-SR04 Sonar"

gpio.setup(trig,gpio.OUT)
gpio.setup(echo,gpio.IN)

while True:
	gpio.set(trig,0)
	print "Measure..."

	gpio.set(trig,1)
	time.sleep(0.00001)
	gpio.set(trig,0)

	pulse_start = time.time()
	while gpio.read(echo)==0:
		pulse_start = time.time()

	pulse_end = time.time()
	while gpio.read(echo)==1:
		pulse_end = time.time()

	pulse_duration = pulse_end - pulse_start
	distance = pulse_duration * 17150
	distance = round(distance, 2)

	print "Distance" , distance `

jreindel, 2019-04-17:
I think @ikoc is right on the issue.
I think @VincentDuf 's issue was that you were using the wrong numbers. 
You need to use the sysfs reference numbers. The Jetson TX2 has a lot more pins than you see on the GPIO header of the development board. The reference numbers refer to the pins out of the actual Jetson TX2 module. If you have ever taken it off, you can see how many connections there are on the bottom (Arrow Electronics site has a good shot of that https://www.arrow.com/en/products/900-82180-0001-000/nvidia). The large number of connections attribute to the large sysfs numbers used. These are similar to the BCM numbers for RaspberryPi, which refer to the physical pin on the Broadcom processor chip.

Gadgetoid, 2021-11-03:
I would welcome the submission of examples- perhaps in an "examples" dir of this repository- to demonstrate use on various platforms and help get users off to a running start. In the interim I'll tag this issue up as a "notice" and leave open for future users.