vitiral/gpio #20: Looking for a new maintainer

vitiral, 2020-09-30:
So I haven't used this library in ~5 years, and haven't really even used python in ~2 years. Every time I need to update PyPi it is a headache for me.

With a million other projects including a full time job, father of 3 including a new infant and the effort I am doing with civboot.org, I will not be able to push updates to this project. If someone is willing to maintain it, please come forward.

Thanks!

Gadgetoid, 2021-08-09:
Coming in blind to this project on a groggy Monday morning without having first reviewed the code I’d like to tentatively volunteer myself.

I maintain about 100 Python libraries intended for the Rasperry Pi and written to support a variety of sensors and expansion boards (HATs). Right now the very Pi-specific “RPi.GPIO” underpins the GPIO functionality of these libraries and limits them to the Pi.

I’ve been interested in libgpio for a while, but it hasn’t quite been ready for prime-time and uptake is slow. I’ve been reliably informed that libgpio2.0 will be riding onto the scene on Debian bullseye so it’s probably time to kick the tires of this library. Where suitable I can then look into porting my libraries over to fast-track adoption.

(congrats on the new infant, I had to fend my youngest off - now 2 - to type this 🤣)

vitiral, 2021-08-09:
Sounds good to me! If you know the process for transferring a pypi project, please let me know...

vitiral, 2021-08-09:
I added you as a collaborator for now.

Gadgetoid, 2021-08-09:
Thank you!

If you log into pypi you can jump straight to here - https://pypi.org/manage/project/gpio/collaboration/

And add "pimoroni" (https://pypi.org/user/pimoroni/) as a collaborator.

Gadgetoid, 2021-08-12:
Thank you! It might take me some time to get up to speed, and I'm also pulled in far too many directions at the best of times, but let's give it a shot.

Gadgetoid, 2021-11-03:
I guess this can be closed for now!

closed 2021-11-03 by Gadgetoid