vitiral/gpio #23: Different path to gpio on Atmel Sama5d3x

GauiStori, 2021-09-06:
Hi

Thanks for the library. I had to change it a bit to make it work on my board, Atmel sama5d3 xplained with kernel 5.4.

To enable GPIO E14 as an output I do:
`echo 142 > /sys/class/gpio/export`
then 
`/sys/class/gpio/pioE14 `
appears but not
`/sys/class/gpio/gpio142`

The only reference to this naming is in:
https://www.at91.com/viewtopic.php?t=23943

I don't know if this is Atmel specific behavior since all other info I find refer to 
`/sys/class/gpio/gpio142`
so the user may have to be able to select format.
The patch below works nicely on my Atmel board

Regards
Gudjon

 ```
@@ -27,7 +27,13 @@ path = os.path
 pjoin = os.path.join
 
gpio_root = '/sys/class/gpio'
-gpiopath = lambda pin: os.path.join(gpio_root, 'gpio{0}'.format(pin))
+#gpiopath = lambda pin: os.path.join(gpio_root, 'gpio{0}'.format(pin))
+def gpiopath(pin):
+    if pin < 0:
+        return False
+    l = ['pioA','pioB','pioC','pioD','pioE']
+    return os.path.join(gpio_root,"%s%d"%(l[pin>>5],(pin&0x1F)))
+
 _export_lock = threading.Lock()
 
 _pyset = set

```

vitiral, 2021-09-07:
That is... pretty strange behavior. At first I thought maybe it was using hexadecimal numbers, but your "fix" wouldn't work for that case, so I'm not sure.

I think the right way to do this would be to have an exposed function for changing the implementation of `gpiopath`, maybe with a few possibilities provided.

Gadgetoid, 2021-09-08:
Looks like the GPIO banks are the upper 3 bits of the GPIO number, and it's N banks of 32 IO pins.

I've never seen this setup before, but I guess it could be called "banked GPIO."

It was my understanding that the kernel driver was supposed to flatten this out to a single consistent numbered set of pins since GPIO banks are a largely irrelevant hardware implementation detail (because your GPIO registers on a 32-bit system can only be 32-pins wide).

I can't find any reference for how this should work, and I can't find the source for the Sama5d3x driver that implements this... oddity. As such I don't know if we can expect any consistency across platforms with banked GPIO- ie: are banks always `A, B, C, D, E, F` etc and are pin numbers always expressed in hex and is it always `pioE14`.

A GitHub-wide search suggests this isn't an Atmel-only pattern- https://github.com/search?q=%2Fsys%2Fclass%2Fgpio%2Fpio&type=code

But then here's the exception to prove the rule - https://github.com/ec-jrc/airsenseur-sensorshost/blob/65a039d8cf0d93f2ccbb8d23abf21c33a10c4b08/Software/HostBoard/Linux/custom/Scripts/usr/local/airsenseur/init_pio#L33-L36

And a similar pattern - https://github.com/chuffman93/cdh-fsw/blob/b56ec0cd6c40b882e03ae78aab0be7868881ce6d/src/servers/CDHStdTasks.cpp#L192-L201

So we have:

1. Banked GPIO
2. Format of `/sys/class/gpio/pio<bank><pin>`
3. Where `<pin>` can be either hexadecimal or decimal

Which *could* be supported in my incoming PR (https://github.com/vitiral/gpio/pull/22) using the class-based approach.

For example:

```
import gpio

pin = gpio.GPIOPin(14, gpio.OUT, banked=True, mode=gpio.HEX)
```

Or, perhaps, a global state set in the library:

```
import gpio
gpio.configure(banked=True, mode=gpio.HEX)

pin = gpio.GPIOPin(14, gpio.OUT)
```

With a view to identifying if it's possible to auto-detect this and have the library configure automatically.

But we don't know:

1. If there are any other formats for `/sys/class/gpio/pio` (maybe offer `prefix=` in `configure`)
2. If there are any other supported bank sizes... ie: for 16 (unlikely) or 64bit registers
3. If the method for unpacking a number into a bank/GPIO is consistent

GauiStori, 2021-09-11:
Hi Philip

Thanks a lot for the answer. I find all the examples consistent with the link to at91 above
https://www.at91.com/viewtopic.php?t=23943
The banks are numbered A to F and the pin numbers 0 to 31 decimal.

I will test your new code ASAP.

Gadgetoid, 2021-11-03:
There's no support for it yet, but I'm making up banked GPIO as a potential enhancement to the library. I'd like to get 1.0.0 released first and build upon that- time provided!

Gadgetoid, 2021-11-03:
What happens if you- for example - `echo E14 > /sys/class/gpio/export`.

Does that fail, or successfully export `/sys/class/gpio/pioE14`?