vitiral/gpio #24: pypi package contains debug enabled line which is creating unwanted debug noise also for other modules than this one

Hadatko, 2021-11-03:
Hi, 
In pypi the package code contains: 
[image]

It would be nice to release same code as in this repository. @Gadgetoid I read that you are maintainer of pypi package. Hope it is still true ;) 

Btw github should have some feature to push new package to pypi after release.

Thank you very much.

vitiral, 2021-11-03:
I think this should be fixed in later versions. Can you simply upgrade?

Hadatko, 2021-11-03:
I am on latest package ;) Here it is ok, but on pypi it is not ;)

Hadatko, 2021-11-03:
For verifying you can download package by yourself and edit gpio python core file https://pypi.org/project/gpio/0.3.0/

Gadgetoid, 2021-11-03:
Next verison is pending in https://github.com/vitiral/gpio/pull/22 but hasn't really had enough eyes on it to move forward just yet.

Could you try grabbing this branch and installing it locally to see if it:
1. Fixes your issue
2. Still works as expected

```
git clone https://github.com/Gadgetoid/gpio -b gadgetoid-working
cd gpio
python3 setup.py install
```

Vary your install comamnd as per your Python version/sudo requirements and/or add `--user` to install for a single user.

Gadgetoid, 2021-11-03:
In fact - if you're willing - you can help move the new version another step closer to release by installing it from test pypi here: https://test.pypi.org/project/gpio-test/1.0.0/

Note: I had to rename the library from gpio to gpio-test (I don't have permissions to publish to test pypi yet), but it will still install a "gpio" module.

If this version works as expected, I'm happy to push it up to pypi and we can move forward from there.

VyCOSystems, 2023-02-13:
> Next verison is pending in #22 but hasn't really had enough eyes on it to move forward just yet.
> 
> Could you try grabbing this branch and installing it locally to see if it:
> 
> 1. Fixes your issue
> 2. Still works as expected
> 
> ```
> git clone https://github.com/Gadgetoid/gpio -b gadgetoid-working
> cd gpio
> python3 setup.py install
> ```
> 
> Vary your install command as per your Python version/sudo requirements and/or add `--user` to install for a single user.

hi I'm confirming this fixed the DEBUG messages issue for me.

Gadgetoid, 2023-02-13:
I've just pushed up v1.0.0. I think I prepped this a while ago but ran into trouble pushing up to testpypi (I don't have permission to upload there).

Any and all feedback appreciated.

VyCOSystems, 2023-02-13:
> I've just pushed up v1.0.0. I think I prepped this a while ago but ran into trouble pushing up to testpypi (I don't have permission to upload there).
> 
> Any and all feedback appreciated.

I installed the latest v1.0.0 using pip and it's working well.
thank you :-)

closed 2023-09-27 by Gadgetoid