vitiral/gpio #30: libgpiod and the future of gpio in Python

Gadgetoid, 2023-09-27:
First off, it's abundantly clear to me that - despite its deprecation - Linux sysfs GPIO (the way this library currently works) is going to stick around for quite some time, and still has uses on embedded platforms with behind-the-curve kernels. I don't want to take any functionality away from people.

However, Linux's new libgpiod is the clear path forward, and the current state of libgpio for Python users is absolutely heinous.

We have:
* The excellent but confusingly named `python3-gpiod`, which is a pure-python re-implementation of `python3-libgpiod`. It works, but is severely lacking in documentation and does not in any way attempt to be "pythonic" in its implementation- being a 1:1 mapping of the reference design. It's available via `pypi` but not (afaik) in any Linux package managers.
* The, effectively a reference design, `python3-libgpio`, this is a C library that is available via - for example - the `python3-libgpiod` Debian package. It does not have a pypi counterpart, so it cannot be specified as a dependency or installed in a virtual env.
* A mystery-meat `libgpiod` package which has been uploaded to `pypi` with no details or indication as to which of the above it resembles
* Third-party libraries such as `RPi.GPIO` (RPi only), `wiringpi` (defunct), and `lgpio` (too broad)
* Libraries like `blinka` or `gpiozero` which attempt to solve the problem of the GPIO ecosystem by sweeping it all under a more Pythonic, user-friendly API. These are inevitably too broad in scope to be a reasonable dependency for- for example- handling an interrupt pin in a simple sensor driver.
* Libraries like this one, which - in my humble opinion - represent a reasonable, Pythonic answer to the problem of GPIO, but rely on a deprecated interface.

To confound this issue, in the case of the low-level libraries (`gpiod` and `libgpio`) documentation is scant.

I first picked up `gpio` (this library) because I wanted to solve this problem, and had (ha ha) mistaken this library for a libgpiod based library.

I still want to solve this problem, and - given how complicated, cryptic and confusing the various `gpio` package names are (`lgpio`, `libgpio` and `libgpio` etc) - making `gpio` (this library) the *canonical* library for Linux GPIO in Python seems like the prudent move. That means adding support for `libgpiod` based GPIO access into this library, and doing it right.

And, thus, this is an open call for feedback and input on this idea, how to approach it and what GPIO should look like in Python.

vitiral, 2023-09-27:
https://stackoverflow.com/questions/74352978 looks like a good reference

Gadgetoid, 2023-09-27:
Okay, plot twist- it's *the author of libgpiod* who uploaded libgpiod to Pypi. The package appears to be irreconcilably broken on Raspberry Pi OS (Debian Bookworm) due to a mismatch in assumed libgpiod versions.

The big problem with no 2 this is that `gpiod` cannot be imported from system packages while in a virtual context, unless the user specifies `--system-site-packages`. I'm trying to figure out how that can be fixed, since depending upon *the* `python3-libgpiod` (whether it comes via apt or pypi) seems to be most reasonable solution.

I suspect we could probably probe for the existence of `/sys/class/gpio/gpiochipX` and, if the library is available, prefer it.

I was broadly hoping that a widely available, widely compatible `python3-libgpiod` library would let us simply specify it as a dependency but it seems to be repeating the same mistakes made by `smbus` (that prompted the pure-python `smbus2`).

We might have to depend upon or largely re-implement the pure-Python alternative: https://github.com/hhk7734/python3-gpiod

Gadgetoid, 2023-09-27:
Note: the linked stackoverflow is, afaik, already out of date.

Gadgetoid, 2023-09-28:
Okay I've got a prototype libgpiod package that I can install (via pip) on a Raspberry Pi.  It - with any luck - works elsewhere too, but I'll admit I'm probing in the dark a little here.

https://github.com/Gadgetoid/libgpiod-python/releases/tag/2.0.0

I'm in the process of working out how this package could become the canonical libgpiod Python distribution, but I realize "works for me" is a very long way from clearing the bar for such an important package.

vitiral, 2023-09-28:
If you can get a gpiod installed from pip, then couldn't users simply use that?

Gadgetoid, 2023-09-28:
> If you can get a gpiod installed from pip, then couldn't users simply use that?

In theory, yes, but in practise it's another API to learn and understand to accomplish the same task. And not a terribly user-friendly one at that.

Perhaps it's hubris thinking it could be better. But it's had multiple years to make a dent in the Python ecosystem and all we've got is a bunch of stackoverflow posts with various snapshots of out of date code.

In all cases, though, it would be *faster* to use the library directly and that's always an option, but I very much believe there's a neat little gap between libgpiod and gpiozero where a very concise GPIO library can sit for the benefit of people who aren't either engineers or kids/beginners respectively.

Also gives a migration path for those using this library to transition painlessly over to libgpiod if/when they get rugged by their distro 😆

vitiral, 2023-09-28:
This does sound concerning, but I don't think that making _this_ library more complicated is going to help anyone. In fact, folks are better helped by this library being stable and constant IMO.

PyPi's fragmented ecosystem is outside the scope of this package :smile: